AlbrechtL/openwrt-docker #1: opkg? How do I add that?

ForDefault, 2024-04-13:
I am confused, you said:

> OpenWrt doesn't provide a user installed package update mechanism all required packages needs to be included into the OpenWrt rootfs image. This Docker images adds the following software to the OpenWrt rootfs:

but [opkg](https://openwrt.org/docs/guide-user/additional-software/opkg) is the package manager and for some reason was left out. 

Could you tell me where I add this in the scripts so I can make a build with it?

AlbrechtL, 2024-04-17:
Inside OpenWrt the `opkg` command is available (see screenshot).

[image]

Please verify it on your system.

ForDefault, 2024-04-21:
Just wanted to say **this project is pretty awesome and glad you achieved this**:

Probably because I am trying to go straight through docker? 
```
docker exec -it openwrt /bin/ash
/ # opkg update
/bin/ash: opkg: not found
/ # opkg update
/bin/ash: opkg: not found
/ # opkg 
/bin/ash: opkg: not found
/ # ls
bin              etc              lib              mkdir            opt              root             sbin             storage          sys              usr
dev              home             media            mnt              proc             run              srv              supervisord.pid  tmp              var
```

My intended use case is not directly connected to the main scope of this repo though. I just wanted an openwrt container running in qemu (needed for wg) on a guest docker network so I could use it with my containers. Your project is the only one that comes close and its been a nightmare trying to reverse engineer your work (fairly extensive lol). If you have any recommendations on how to do that I would appreciate it. If not I will close this out next time I check! 
Thank you again.

closed 2024-04-21 by ForDefault

AlbrechtL, 2024-04-21:
Important is to know that inside the container OpenWrt is running as virtual machine. So `opkg` is not  available inside the container rootfs. So `docker exec -it openwrt /bin/ash ` is not working.

To access OpenWrt the rootfs you have two options.
1. Access via the web VNC console (see to picture above) 
2. Access via SSH (depending on network configuration and OpenWrt firewall configuration)