AlbrechtL/openwrt-docker #5: Request: Add User Mode Networking

lars18th, 2024-10-01:
Hi,

This project is very interesting. However, for developing the execution in Docker Desktop (Windows) will be desirable. At time, the networking is implemented using only the `macvlan` support. Other similar projects (using QEMU/KVM inside Docker) are supporting as an alternative the User Mode Networking. See more here: https://wiki.qemu.org/Documentation/Networking

I hope you want to consider it.

AlbrechtL, 2024-10-02:
Actually, user mode network is already included. If you set the environment variable to `WAN_IF: "host"` is uses SLIRP. But the performance is really bad.

What is your use case?

lars18th, 2024-10-02:
Hi @AlbrechtL ,

> What is your use case?

Running it (for testing and developing) in Windows 11 (Docker Desktop + WSL).

I have shown (a few minutes ago) that yes the SLIRP mode is supported...
But I suggest to do this small changes to make it more visible:
- In `docker-compose.yml` add something like: `#If WAN_IF is missing then "host" mode is selected`.
- Do the same for `LAN_IF` and add support for `host` mode in: https://github.com/AlbrechtL/openwrt-docker/blob/9fc6feef696ea19a7e8a16d55a4ec2856086d739/src/run_openwrt.sh#L151

Thank you.

AlbrechtL, 2024-10-02:
> In docker-compose.yml add something like: #If WAN_IF is missing then "host" mode is selected.

Great idea. I will add it.

> Do the same for `LAN_IF` and add support for `host` mode in: 

I did some experiments with it. The thing is that OpenWrt configures a DHCP server at the LAN side because it is a router. The SLIRP implementation on the other side has fixed IPs, a DHCP server and router functionally. See the QEMU SLIRP concept picture.
[image]
Source: https://wiki.qemu.org/Documentation/Networking

So you don't have any benefits of SLIRP at the LAN side. What do you want to do with it? Maybe I miss something here.

lars18th, 2024-10-02:
> What do you want to do with it? Maybe I miss something here.

Do tests of complex configurations with it in my workstation.

Now it runs inside Windows (Docker Destop + WSL2) with this config:
((  NOTE: in `.wslconfig` >> `[wsl2] networkingMode=mirrored`  ))
```
    environment:
        WAN_IF: "host"
        #LAN_IF: "" # Not defined. In the future "host"

    volumes:
      - ./data:/storage/  # With this you control where is the storage... start docker from the wsl shell.
```
Then the `localhost:8006` is the program GUI. And with `/usr/bin/fw_wan_open_*` you will enable `localhost:8000` for Luci and `localhost:8022` for SSH.

Nice!

lars18th, 2024-10-02:
Hi @AlbrechtL ,

One small suggestion for the `"host"` mode of `WAN_IF` and `LAN_IF`:
- With this "host mode networking" it will be interesting to add support to select IPv4 and/or IPv6 (with the option `ipv4=on|off and ipv6=on|off`). Using IPv6 it's a pain in several environments. Therefore if you support something like `host4` to use `ipv4=on,ipv6=off` then it will be useful. Options could be `host` (default 4 and 6), `host4` (only 4) and `host6` (only 6). The idea is to block traffic at the qemu virtual interface and after configure OpenWRT with the corresponding values.

I hope you agree with that suggestion.

AlbrechtL, 2024-10-02:
I'm not sure if I understood you correctly. But you are free to implement the options that you are needing, test it and send a pull request afterwards.

lars18th, 2024-10-02:
> I'm not sure if I understood you correctly. But you are free to implement the options that you are needing, test it and send a pull request afterwards.

OK. But the problem is that the current documention is not sufficient for some aspects (however the project seems to be good maintained). Any idea to execute the tests?

AlbrechtL, 2024-10-02:
I just realized that I already implement SLIRP in LAN. The environment variable needs to be cleared, see
https://github.com/AlbrechtL/openwrt-docker/blob/1e47d770f4b7b7328525238e0f14cadf788191cb/src/run_openwrt.sh#L150-L152

But I never used it. Fell free to play with it.

> Any recommendation to execute the tests inside the command line?

A good starting point is to build the Docker image yourself. Executing the automated tests something for an advanced usage. But you can read the test pipline https://github.com/AlbrechtL/openwrt-docker/blob/master/.github/workflows/test.yml how to do it.

lars18th, 2024-10-02:
Hi @AlbrechtL ,

I created the PR #7 to make it explicit. It's relevant in environments like Windows where only user networking is available. It works without problems with `LAN_IF="host"` and `WAN_IF="host"`.

AlbrechtL, 2024-10-02:
Thanks for your work! I merged your PR.

closed 2024-10-02 by AlbrechtL