AlbrechtL/openwrt-docker #6: Suggestions: Small UI improvements

lars18th, 2024-10-02:
Hi @AlbrechtL ,

I've these suggestions after starting to use this project to improve the UI experience:
- In the "Get Continer" (or "Get Openwrt") I want to see the QEMU parameters. The full command-line will be useful to detect some troubles.
- When starting the openwrt it will be interesting to print in the container log (I run it from the command line) something like `Guest: OpenWRT 23.05.5 running`. The problem is that you see (last): `Booting image using QEMU emulator version 9.0.2...`. And the you can't trace the status from the command line. Almost a line will be useful.
- In the Web-UI the button for "Open Luci Web" could be two buttons: One for the LAN and one for the WAN. At time, it's more easy (with user networking) to open SSH and Luci on WAN. Therefore the button could open `http://127.0.0.1:8000` too. And please, use explicit `http://` because by default `https` is not enabled in Luci.

I'll provice more feedback.
Regards.

AlbrechtL, 2024-10-02:
Thanks for your thoughts. When I come back to the web interface, I will consider your ideas.

lars18th, 2025-01-12:
Hi @AlbrechtL ,

After thinking about a KISS solution for the "Open Luci Web" button, I've this simple solution:
- Add in the `docker-compose.yml` file this new parameter: `WEB_TWEAK_Luci_Web: "http://localhost:8080"`.

So, then the user will have the option of rewriting the value of this link. It will be his responsability. But at the same time it will easy to edit it.

What you think about that?

AlbrechtL, 2025-03-28:
Some UI progress.

To QEMU parameters: I added it in https://github.com/AlbrechtL/openwrt-docker/commit/4a1abe89c31f0b064048ffe154faeee1ea9b8d0a
[image]

AlbrechtL, 2025-03-29:
> After thinking about a KISS solution for the "Open Luci Web" button, I've this simple solution:

Thanks for this great idea. Added in https://github.com/AlbrechtL/openwrt-docker/commit/9a736b5cd61033b72153ec61aa546421f451dedc.

It would be really nice if you could test it.
https://github.com/AlbrechtL/openwrt-docker/blob/9a736b5cd61033b72153ec61aa546421f451dedc/docker-compose.yml#L42-L46

AlbrechtL, 2025-03-29:
I'm closing this issue for now.

closed 2025-03-29 by AlbrechtL