AlbrechtL/openwrt-docker #8: Local IP host

qwdsfsfvss, 2024-10-15:
Hi,
I'm trying to use your docker's image on a x86 board with dual ethernet.
The Idea is to configure one ethernet as WAN and the other one as a LAN.
Following your compose fiel I configure LAN and WAN directly on machine network interface.
After on OpenWRT, I configured WAN as a dhcp client, and LAN with static address with dhcp server enabled.
If I connect something on LAN port it receives correctly the ip address from dhcp server on OpenWRT.

My problem is on how to give to the host machine an IP address of that LAN so the can talk.

Thank you for your work!

AlbrechtL, 2024-10-21:
> My problem is on how to give an IP address of the LAN to the host machine, otherwise the host is cutted off.

Currently, this is not possible. Both WAN and LAN Ethernet interfaces are added to OpenWrt exclusively. The only option you have is to add a third Ethernet interface e.g. with a USB-Ethernet adapter to have access to the host.

qwdsfsfvss, 2024-10-23:
Sorry,
I want the host that can reach client connected on Openwrt LAN

AlbrechtL, 2024-10-23:
You have two options:
1. As mentioned above add a third physical Ethernet interface and connect the LAN OpenWrt Ethernet interface and the host Ethernet interface via an external Ethernet switch together.
2. Use the `veth` options to create a virtual LAN Ethernet interface at host side. Then create a Ethernet bridge at the host side and add the virtual OpenWrt Ethernet interface and the host physical Ethernet interface to this bridge. Details about that you can find at https://developers.redhat.com/blog/2018/10/22/introduction-to-linux-interfaces-for-virtual-networking or https://wiki.archlinux.org/title/Network_bridge.

qwdsfsfvss, 2024-10-23:
Thank you for your answer.
I' ve tried your second option, but when I use veth as LAN on Openwrt and I change the subnet of LAN in something like 192.168.xx.xx/24 the veth interface remains on the old address. It is like the address of veth is hardcoded.
Thank you

AlbrechtL, 2024-10-24:
Yes you are right. In `run_openwrt.sh` the IP address is hard-coded. See: https://github.com/AlbrechtL/openwrt-docker/blob/2b2910eb2b95689a21d777104be9523f86705d3c/src/run_openwrt.sh#L72

1. Nevertheless the hard-coded address does option 2 solves your issue?
2. What would be a better approach in your option?

qwdsfsfvss, 2024-10-24:
Hi, 
thank you for your time.
I will try your solution.
If possible, I think it will be better that Openwrt will configure veth, since veth is the LAN of OpenWRT.
Thank you very much

AlbrechtL, 2024-10-24:
Just to be clear `LAN_IF: "veth,noautoip"` is not implemented, currently.

AlbrechtL, 2024-10-28:
In https://github.com/AlbrechtL/openwrt-docker/commit/aaab668d70e189620bdf2915e039bfd4eaf04761 I added the option `nofixedip`. See docker-compose.yml for details
https://github.com/AlbrechtL/openwrt-docker/blob/aaab668d70e189620bdf2915e039bfd4eaf04761/docker-compose.yml#L28

I hope that with this new feature, you can work.

closed 2024-11-01 by AlbrechtL

Switzerg95, 2025-04-17:
I added a usb ethernet interface and created a bridge with that interface and the default lan veth.

client behind usb dongle can reach internet, gets a dhcp address from the openwrt docker container, but it cant talk to anything else thats on some of my mesh nodes, but any mesh node LAN members (same l2) can reach each other fine. 

Is there a different suggestion on how we could make this work?

AlbrechtL, 2025-04-17:
@Switzerg95 Please move your question to https://github.com/AlbrechtL/openwrt-docker/discussions. From a gut feeling, I would say that it could be something with the firewall. But I'm not sure if I understood your issue correctly. So, please provide a network topology sketch.

AlbrechtL, 2025-04-18:
@Switzerg95 Thanks for https://github.com/AlbrechtL/openwrt-docker/discussions/36. Let's continue there.