AlbrechtL/openwrt-docker #11: x86-64 failure

8jef, 2024-12-12:
Hello,

I've tried your docker image twice, first in podman, then in docker, it will start an fail within a second.

System used: 
pop!_os 22.04 (ubuntu)
x64 AMD Ryzen 2400G

This is the log from docker:

2024-12-12 17:54:47 3.21.0
2024-12-12 17:54:47 NAME="Alpine Linux"
2024-12-12 17:54:47 ID=alpine
2024-12-12 17:54:47 VERSION_ID=3.21.0
2024-12-12 17:54:47 PRETTY_NAME="Alpine Linux v3.21"
2024-12-12 17:54:47 HOME_URL="https://alpinelinux.org/"
2024-12-12 17:54:47 BUG_REPORT_URL="https://gitlab.alpinelinux.org/alpine/aports/-/issues"
2024-12-12 17:54:47 **** End Alpine release information ****
2024-12-12 17:54:47 CPU architecture: 'x86_64'
2024-12-12 17:54:47 ❯ Detected generic container
2024-12-12 17:54:47 *******************************************
2024-12-12 17:54:47 qemu-openwrt-web-backend proof-of-concept
2024-12-12 17:54:47 *******************************************
2024-12-12 17:54:47 ❯ /storage/squashfs-combined-1c026911-f23f-4b74-8925-21c4c09df041.img exists. Nothing to do.
2024-12-12 17:54:47 ❯ squashfs-combined-1c026911-f23f-4b74-8925-21c4c09df041.img is up to date. Nothing to do.
2024-12-12 17:54:47 ❯ Checking for KVM ...
2024-12-12 17:54:47 ❯ Booting image using QEMU emulator version 9.1.2...
2024-12-12 17:54:47 Could not access KVM kernel module: No such file or directory
2024-12-12 17:54:47 qemu-system-x86_64: failed to initialize kvm: No such file or directory
2024-12-12 17:54:47 multirun: one or more of the provided commands ended abnormally

Would love to try and use if you can fix it. Thanks!

AlbrechtL, 2024-12-15:
It seems that you don't have KVM activated. See log `"2024-12-12 17:54:47 Could not access KVM kernel module: No such file or directory"`. 

KVM (kernel virtual machine) is mandatory. 

Please check how you can install/activate it under you OS.

AlbrechtL, 2024-12-26:
Closed because of inactivity.

closed 2024-12-26 by AlbrechtL