AlbrechtL/openwrt-docker #13: port 9000 in use

benisai, 2025-01-05:
openwrt    | *******************************************
openwrt    | qemu-openwrt-web-backend proof-of-concept
openwrt    | *******************************************
openwrt    | ❯ /storage/squashfs-combined-9331e7e7-1960-4be0-bafc-5f28847dbbc1.img exists. Nothing to do.
openwrt    | ❯ squashfs-combined-9331e7e7-1960-4be0-bafc-5f28847dbbc1.img is up to date. Nothing to do.
openwrt    | 2025/01/05 16:45:06 [emerg] 4047991#4047991: bind() to 0.0.0.0:9000 failed (98: Address in use)
openwrt    | 2025/01/05 16:45:06 [notice] 4047991#4047991: try again to bind() after 500ms
openwrt    | unable to initialize usb specnginx: [emerg] bind() to 0.0.0.0:9000 failed (98: Address in use)
openwrt    | nginx: [warn] conflicting server name "" on 0.0.0.0:80, ignored
openwrt    | nginx: [warn] conflicting server name "" on [::]:80, ignored
openwrt    | ❯ Checking for KVM ...
openwrt    | ❯ KVM detected
openwrt    | ❯ Creating virtual Ethernet interfaces pairs between host system (veth-openwrt0) and container (veth1)...
openwrt    | ❯ Booting image using QEMU emulator version 9.1.2...
openwrt    | nginx: [emerg] bind() to 0.0.0.0:9000 failed (98: Address in use)
openwrt    | 2025/01/05 16:45:06 [emerg] 4047991#4047991: bind() to 0.0.0.0:9000 failed (98: Address in use)
openwrt    | 2025/01/05 16:45:06 [notice] 4047991#4047991: try again to bind() after 500ms
openwrt    | nginx: [emerg] bind() to 0.0.0.0:9000 failed (98: Address in use)
openwrt    | 2025/01/05 16:45:06 [emerg] 4047991#4047991: bind() to 0.0.0.0:9000 failed (98: Address in use)
openwrt    | 2025/01/05 16:45:06 [notice] 4047991#4047991: try again to bind() after 500ms
openwrt    | 2025/01/05 16:45:06 [emerg] 4047991#4047991: bind() to 0.0.0.0:9000 failed (98: Address in use)
openwrt    | 2025/01/05 16:45:06 [notice] 4047991#4047991: try again to bind() after 500ms
openwrt    | nginx: [emerg] bind() to 0.0.0.0:9000 failed (98: Address in use)
openwrt    | 2025/01/05 16:45:06 [emerg] 4047991#4047991: bind() to 0.0.0.0:9000 failed (98: Address in use)
openwrt    | 2025/01/05 16:45:06 [notice] 4047991#4047991: try again to bind() after 500ms
openwrt    | nginx: [emerg] bind() to 0.0.0.0:9000 failed (98: Address in use)
openwrt    | nginx: [emerg] still could not bind()
openwrt    | 2025/01/05 16:45:06 [emerg] 4047991#4047991: still could not bind()
openwrt    | qemu-system-x86_64: terminating on signal 15 from pid 4047939 (multirun)
openwrt    | multirun: one or more of the provided commands ended abnormally

benisai, 2025-01-05:
I'm trying to get into luci, but cant. I think this is why :(

AlbrechtL, 2025-01-07:
You are right! On your host system, port 9000 is in use by another service.

Unfortunately, port 9000 is hard coded currently. It is possible to move your other service to another port?

benisai, 2025-01-07:
This is not on my host side. This is within the container.

AlbrechtL, 2025-01-07:
OK. I think I need to explain something more in detail.

Indeed, the error message comes from the container. Despite other services, the Docker runtime is not reserving port 9000. That's why the container is starting even when port 9000 is not available at the host.

Instead, a reserve proxy is started from the container direct at the host, see https://github.com/AlbrechtL/openwrt-docker/blob/d61dcf512c42ada74028510c936449c34f8730c6/src/init_container.sh#L65. Simply spoken: This is a container escape and is necessary to make LuCI available outside the container and qemu.

The nginx configuration you can find [here](https://github.com/AlbrechtL/openwrt-docker/blob/]d61dcf512c42ada74028510c936449c34f8730c6/web-frontend/nginx-luci.conf).

closed 2025-01-11 by AlbrechtL

benisai, 2025-02-02:
So how do I expose the webui via my VPS? I think the built in reverse proxy is overly complex and making this a non standard setup with docker-compose :)

benisai, 2025-02-02:
why is there a built-in proxy btw? I think its over arch'd. the ttyl should have its port exposed to the container, and the luci should have its own port, no reverse proxy needed within the container.

benisai, 2025-02-02:
What port do I need to expose in docker-compose to get to the luci interface? I'm really confused by the built-in nginx RP .

benisai, 2025-02-02:
I figured it out - 8000:8000