AlbrechtL/openwrt-docker #15: Question:

Switzerg95, 2025-01-07:
Does this implementation support PCI passthrough in some fashion? I'm able to get usb wifi pass through working great but im not able to figure out why i cant get my intel 9260 pci adapter seen in lspci in the openwrt docker instance.

AlbrechtL, 2025-01-07:
The current implementation doesn't support PCI pass-through. That's why you don't see your 9260 Wi-Fi device in OpenWrt.

Generally, `qemu` support PCI pass-through (see https://gist.github.com/lisovy/1f737b1db2af55a153ea). So it should be possible to add this feature somehow. I think this feature could be interesting.

Pull requests are welcome!

Switzerg95, 2025-01-07:
That makes sense. Was spending alot of time on this and ended up hitting a wall.
Thank you!
This would be a great enhancement.

Switzerg95, 2025-01-07:
How would you go about this?

Switzerg95, 2025-01-10:
Thanks ! This is exciting.

I will check this soon.

Switzerg95, 2025-01-10:
When attempting to re-build with the new image it fails to update opkg and hangs forever.

AlbrechtL, 2025-01-10:
> When attempting to re-build with the new image it fails to update opkg and hangs forever.

Just try it again.

But I recommend using the image from Docker hub `albrechtloh/openwrt-docker:latest`. Look to [docker-compose.yml](https://github.com/AlbrechtL/openwrt-docker/blob/2bb5a119fbb6b6010bbd5c37ad1261a95f464b3c/docker-compose.yml).

AlbrechtL, 2025-01-11:
I'm closing this issue because PCI pass-through is supported. If you need more help feel free to start a new discussion here: https://github.com/AlbrechtL/openwrt-docker/discussions/categories/general

closed 2025-01-11 by AlbrechtL

Switzerg95, 2025-01-14:
> Good news. In [2bb5a11](https://github.com/AlbrechtL/openwrt-docker/commit/2bb5a119fbb6b6010bbd5c37ad1261a95f464b3c) I added initial PCI pass-through support.
> 
> Usage:
> 
> https://github.com/AlbrechtL/openwrt-docker/blob/2bb5a119fbb6b6010bbd5c37ad1261a95f464b3c/docker-compose.yml#L37-L44
> 
> Feel free to test it and give feedback, please.

Tested on Ubuntu 22.04.05 LTS with intel ax200 PCIE adapter

[image]

AlbrechtL, 2025-01-20:
Thanks for the good news!

Switzerg95, 2025-02-19:
What about m2 ....?