AlbrechtL/openwrt-docker #16: Confiure firewall from docker-compose.yml

lars18th, 2025-01-07:
Hi @AlbrechtL ,

To setup the container I've found a trouble:
- In the initial setup, in some use cases, you need to go to the console (using the web VNC) to execute the command/s `/usr/bin/fw_wan_open_http` and/or `/usr/bin/fw_wan_open_ssh`.

However, I feel this is wasted time and unnecessary.
Why not provide an environment variable to enable/disable them?
For example:
```
    environment:
      # Enable/disable internal firewall for...
      FIREWALL_HTTP_PASS: "true" # port open, equivalent to run the script  /usr/bin/fw_wan_open_http
      FIREWALL_SSH_PASS: "true"  # port open, equivalent to run the script  /usr/bin/fw_wan_open_ssh
```

I know you might say: Well, LAN interfaces have it. But I'm running with `WAN_IF: "host"` and `LAN_IF: "none"` because running on Windows+WSL2+DockerDesktop I need to pass the LAN interface using a USB device.

What you think?

AlbrechtL, 2025-01-08:
To configure something inside OpenWrt from the `docker-compose.yml` file is not a trivial task because OpenWrt is running as virtual machine.

But it is possible to invoke commands to OpenWrt when OpenWrt is booted via the [QEMU Machine Protocol](https://wiki.qemu.org/Documentation/QMP). Inside this repository is already a QMP shell script [qemu_qmp](https://github.com/AlbrechtL/openwrt-docker/blob/d61dcf512c42ada74028510c936449c34f8730c6/src/qemu_qmp.sh) that could be used.

Your proposed environment variables `FIREWALL_HTTP_PASS` and `FIREWALL_SSH_PASS` are too specific to me. I'm thinking more to have a generic environment variable to run custom commands inside OpenWrt. This command can be used to apply firewall configurations or something else. Security wise people shall know what they are doing because all commands are executed as root without user authentication in OpenWrt.

What is necessary to implement?
1. Add new environment variable e.g. `OPENWRT_AFTER_BOOT_CMD`
2. Detection when OpenWrt is booted successfully (e.g. via QMP polling)
3. Invoke command to OpenWrt

Would that something that could work for you?

lars18th, 2025-01-08:
Hi @AlbrechtL ,

Your proposal for a _generic_ method to run one custom script from the `docker-compose.yml` file is a great idea!

I hope you want to add support for it.
Thank you.

lars18th, 2025-01-08:
Hi @AlbrechtL ,

One suggestion: Why not add support for `OPENWRT_AFTER_BOOT_CMD` and `OPENWRT_AFTER_FIRST_BOOT_CMD` ?
The idea is to run some commands *only* one time. And others at every boot.

You agree with that?

AlbrechtL, 2025-01-08:
>I hope you want to add support for it.

See: https://github.com/AlbrechtL/openwrt-docker/commit/f7725c1a021960e4305fbf65d3f21b78df6ed440

Feel free to test it and give feedback, please.

lars18th, 2025-01-09:
Hi @AlbrechtL ,

Thank you! Tested and working. However, a simple suggestion: Add this example too:
```
#OPENWRT_AFTER_BOOT_CMD: "/usr/bin/fw_wan_open_http; /usr/bin/fw_wan_open_ssh" # Exemple to execute two commands
```

AlbrechtL, 2025-01-09:
> Add this example too:

Great idea, added in https://github.com/AlbrechtL/openwrt-docker/commit/3fb54175d0dcd9a340fa5b9c39c88685066ec8b0.

> Why not add support for OPENWRT_AFTER_BOOT_CMD and OPENWRT_AFTER_FIRST_BOOT_CMD 

To add `OPENWRT_AFTER_FIRST_BOOT_CMD` is even more complicated compared to `OPENWRT_AFTER_BOOT_CMD`. The reason is that it is hard to detected when OpenWrt is booted the first time. During the container image build, OpenWrt is booted as well as when the container is updated. So there is no real first boot in the container. A workaround would be to put the file somewhere in the file system. If the file exists, there is no first boot.

Here is an example how to implement such a behavior with the `OPENWRT_AFTER_BOOT_CMD` variable 
```
OPENWRT_AFTER_BOOT_CMD: '[ -f /etc/first_boot_flag ] || { touch /etc/first_boot_flag; <first_boot_command>; }'
```

I will close this topic for now, because it is resolved.

closed 2025-01-09 by AlbrechtL

lars18th, 2025-01-10:
Hi @AlbrechtL ,

> To add `OPENWRT_AFTER_FIRST_BOOT_CMD` is even more complicated compared to `OPENWRT_AFTER_BOOT_CMD`. The reason is that it is hard to detected when OpenWrt is booted the first time. During the [...]

Right! This is the approach that I'm following.
Thank you.