AlbrechtL/openwrt-docker #20: Suggestion: Add "ports" info insde "Get Container Information"

lars18th, 2025-01-09:
Hi @AlbrechtL ,

I'm missing the redirection (aka _ports_) information in the "Get Container Information" section.

For example, when using the HTTP and/or SSH services _over WAN connection_, the redirection of these ports is very relevant.

Futhermore, related with the request on #6, I suggest to add a secondary button "OPEN OWRT LUCI WEB (WAN)" with:
- the HTTP protocol (and not with HTTPS because it doesn't work, I feel because only 8000->80 is configured).
- the REDIRECTED port (not the default 9000), parsing the redirection of the 8000 port in the container (or using directly the default 8000).

You agree?

AlbrechtL, 2025-01-11:
Thanks for your ideas!

> For example, when using the HTTP and/or SSH services over WAN connection, the redirection of these ports is very relevant.

Which port do you mean? 8000 or 8022?

> Futhermore, related with the request on https://github.com/AlbrechtL/openwrt-docker/issues/6, I suggest to add a secondary button "OPEN OWRT LUCI WEB (WAN)" with:

I'm still thinking how to implement that. Having two buttons is not the solution that I like.

lars18th, 2025-01-11:
> > For example, when using the HTTP and/or SSH services over WAN connection, the redirection of these ports is very relevant.
> 
> Which port do you mean? 8000 or 8022?

Almost the three included in the example file: 8006, 8000 and 8022.
But I feel the best is to print _all_ port listed in the `ports:` section (if this is feasible).

> > Futhermore, related with the request on #6, I suggest to add a secondary button "OPEN OWRT LUCI WEB (WAN)" with:
> 
> I'm still thinking how to implement that. Having two buttons is not the solution that I like.

Ok. Don't worry. I'm sure you'll found a good solution for it. My use case is when the LUCI port is reachable only over the WAN port (with redirection), and not over the LAN port.

Thank you.

AlbrechtL, 2025-01-11:
> Almost the three included in the example file: 8006, 8000 and 8022.
> But I feel the best is to print _all_ port listed in the `ports:` section (if this is feasible).

I think there is no benefit to have these ports inside "Container Information" button. Users explicitly configures the ports and should know it. In the future I will overwork the  "Container Information" page because too much information is shown currently. Showing the ports will add even more information.

lars18th, 2025-01-12:
Hi @AlbrechtL ,

You're the developer of this useful project. However, please take note of some suggestions from the point-of-view of the users:

- The current information inside "Container Information" it's true that it's a lot. But, instead of _removing_ information in the future it could be interesting to provide _two_ view modes: simple mode and advanced mode. For "developers" _all_ information has sense.
- Similar to this, is the current `DEBUG: "y"` inside the `docker-compose.yml`. Why you want to remove this in the future? This info _is_ useful to detect errors. And by default it's off.
- Regarding the redirected `ports` section: it has sense when you run multiple instances of this container. Or when you launch them using some script and the values could be floating. Or when you have multiple sites with different instances that use different values. Please, accept that _information is valuable for different use cases_. Therefore, if you can (with low effort) provide this information, then you're helping the users.

In any case, I don't want to force you to do something undesirable or unnecessary. I just want to recommend you to think about the best way forward in development.

Thank you for your good effort with this project.

AlbrechtL, 2025-01-20:
I got you points! Currently, this feature has not the highest priority. But I keep this issue open so that people can rate for it ;-).