AlbrechtL/openwrt-docker #21: Comment about "missing" LAN_IF

lars18th, 2025-01-10:
Hi @AlbrechtL ,

I suggest to add some info about using a _missing_ `LAN_IF`. This is different that a `LAN_IF: "none"` (that in fact doesn't have sense and is not supported).

And when has sense a _missing_ `LAN_IF` variable? So, for example, when you want to (manually) attach the LAN port to a USB device. Without the `LAN_IF` parameter the `br-lan` bridge is created but no physical device is attached to it. Therefore, you need to attach _internal_ devices to it (aka devices available inside the OpenWRT instance).

I feel this is a relevant information that has sense to add to the `readme.md` and `docker-compose.yml` files.
Do you agree?
Regards.

AlbrechtL, 2025-01-11:
I'm not quite sure if I understood you correctly. If `LAN_IF` is unset the LAN interface is working in host mode, see
https://github.com/AlbrechtL/openwrt-docker/blob/812fa9e4c1cf5c7177f022bbd85448c59ecea383/src/run_openwrt.sh#L200-L201

But you want to have the behavior that no LAN hardware interface is added in that case. Is that right?

lars18th, 2025-01-11:
Oh, that's not my assumption! But in any case, the idea is that you pointed: no _default_ interface attached to the LAN bridge.

AlbrechtL, 2025-01-11:
For some reason the first Ethernet interface in `qemu` will always the LAN interface, the second one is becoming the WAN interface. For that reason it is not possible to have only a WAN interface. Or it least I didn't find a way how to tell OpenWrt only to have a WAN interface. Anyway when you configure the LAN side in `host` mode and WAN side in `none` mode `qemu` will add only one Ethernet interface. It is the business of OpenWrt how to use this interface. Again, from what I saw is that it will be the LAN interface than.

AlbrechtL, 2025-01-11:
In https://github.com/AlbrechtL/openwrt-docker/commit/7a1dabd1362ab2d2d3c5550e83f9e2ca6f4cf59f I improved the [docker-compose.yml](https://github.com/AlbrechtL/openwrt-docker/blob/master/docker-compose.yml) documentation. I hope the things are more clear now.

lars18th, 2025-01-12:
Hi @AlbrechtL ,

Thank for the improvement in the documentation. It's nice! 👍 

However, I suggest to do a bit more. Let me to explain:

- The (basic) use case to support is when the user wants to connect the LAN port a USB device. Now it's easy for the WAN port, because the problem is that OpenWRT wants almost a network device to function as the primary LAN interface. And this has sense because any network device requires a network connection, and the primary port is the LAN (not the WAN).

Therefore, I've this suggestion:

1. First part is add support for `socket` networking. This type of `netdev` in qemu is a simple bidirectional socket to share network flow between two (or more) ends. See more at https://john-millikin.com/improved-unix-socket-networking-in-qemu-7.2
2. The second part is to add support for configuring a `dummy` network device. It could be a TCP listen socket (not connected), a UNIX socket connect to `/dev/null` or any not-connected socket. This then could be assigned as the _primary_ LAN port. And after booting we can configure the USB device to attach it to the `br-lan` where the `dummy` device is connected. And down this `dummy` device. That's in fact a workaround for the use case, but it will work. And it has the advantage of the support for socket networking, that could be useful when running multiple VMs.

What you think about this?

AlbrechtL, 2025-01-20:
About your socket networking idea: It is looking really hacky to me and I don't know what the performance is. I would expect a significant performance lost.

If users are needing only one attached Ethernet interface, I would suggest to simple change the OpenWrt configuration in this case. This could be done via the `OPENWRT_AFTER_BOOT_CMD` option.

closed 2025-01-20 by AlbrechtL