AlbrechtL/openwrt-docker #22: How to write user command from OpenWRT in Docker console?

lars18th, 2025-01-12:
Hi @AlbrechtL ,

I feel this will be already supported, but I don't know how to do it. So please help me:

- At time in the docker console we can see several `openwrt` command like:
```
openwrt  | Detected OpenWrt is booted
openwrt  | Run command "/usr/bin/fw_wan_open_http; /usr/bin/fw_wan_open_ssh"
openwrt  | Command "/usr/bin/fw_wan_open_http; /usr/bin/fw_wan_open_ssh" exit successfully
```
- What I need is to print to this console from _inside_ the OpenWRT instance.

Use case: Print all networking addresses configured/used in the instance.

So, is there a script to achieve this, or is there a `/dev/console` device for this?
Thank you.

AlbrechtL, 2025-01-12:
Currently, it is not possible to write something from OpenWrt into the Docker logs. The only option you have is using `OPENWRT_AFTER_BOOT_CMD` to print some information into the logs.

lars18th, 2025-01-12:
Ok. Thank you!

For me it's sufficient if I could do it with the `OPENWRT_AFTER_BOOT_CMD` command. Because it's easy to call to a _complex_ script inside the OpenWRT instance. So, I can assume that _all_ echos from this script will be writen to the Dockers log?

AlbrechtL, 2025-01-12:
> So, I can assume that all echos from this script will be writen to the Dockers log?

I think so. But the `OPENWRT_AFTER_BOOT_CMD` feature is brand new. So we need to get practical experience.

closed 2025-01-20 by AlbrechtL