AlbrechtL/openwrt-docker #26: Backup config

Mailblocker, 2025-01-27:
Hi,
is there a way to store the config settings like in a docker volume? E.g. store /etc/ of the openwrt to some defined place on the host. I do not know if this is sufficient for openwrt.

AlbrechtL, 2025-01-29:
Unfortunately, that is not possible because OpenWrt is running inside a VM and the Docker container doesn't have direct access to the OpenWrt file system.

Is there a specific reason to have the `etc` directory stored in a volume? The complete OpenWrt filesystem image incl. the `etc` directory is stored in a volume.

closed 2025-01-29 by AlbrechtL

Mailblocker, 2025-01-29:
No specific reason about the '''etc''' directory, I just have no clue about how openwrt works and where it stores all the configuration stuff.

So the way to go is to backup the config via the LuCI web interface? I am asking this because I am testing the setup and might move it to another machine later on. But I don't want to go through all menus again and set everything up step by step.

AlbrechtL, 2025-01-29:
Understood! You have two options:
1. Use native OpenWrt features (like LuCI or `sysupgrade` at the shell)
2. Simply copy the complete openwrt container volume to your new machine.

Mailblocker, 2025-01-29:
Thanks for the quick answer!

Ok, I will store the config via LuCI, setup the container to use a default path via docker `--data-root` on the new machine, load the config and add the new default path to my backup system (automatic backup of the openwrt container/setup).

Mailblocker, 2025-02-02:
I have to ask since I found no solution so far: 
What are the volume names used in this container? So I can relocate them without moving all running containers to another location.

AlbrechtL, 2025-02-04:
> What are the volume names used in this container?

The volume name is defined in the `docker-compose.yml`. 
See: https://github.com/AlbrechtL/openwrt-docker/blob/ca36fa1e193abac21f184e53577d05c636132e88/docker-compose.yml#L148

> So I can relocate them without moving all running containers to another location.

Yes.
* https://www.docker.com/blog/back-up-and-share-docker-volumes-with-this-extension/
* Or just google "docker backup volume"