AlbrechtL/openwrt-docker #29: PCI '0000' to much?

schnellert, 2025-03-04:
Hi,

I wanted to start the OpenWRT container with your setup, I have seen that '0000' is always in front of the PCI value, but mine starts with '0002'. Means it has '0000' to much and therefore cannot find the PCI device.

`docker logs -f openwrt`
[image]

`lspci:`
[image]

schnellert, 2025-03-06:
Hi,

Thanks. After a test, it looks good.

But i think you have to implement the same in the `cleanup_container.sh`, I get something similar.

Again because it starts with `0000`:
[image]

However, the container has stopped because of this:

[image]

AlbrechtL, 2025-03-06:
Oh, I don't know what I did yesterday :-(. The code could not work.

Fixed again in https://github.com/AlbrechtL/openwrt-docker/commit/277fb90577f75d33045e8a3d263dcc2b5c4b5791.

I would be really nice if you could test it again when the build ended successful.

schnellert, 2025-03-06:
Okay, the problems with PCI have been solved. 

But the container doesn't actually come up, should I open a new issue for this or can it be done "quickly"? 

[image]

Thank you very much.

AlbrechtL, 2025-03-06:
Cool that it's working. I will close this issue.

Regarding your last report. You ran into the port 9000 issue. See https://github.com/AlbrechtL/openwrt-docker/issues/13) for details.

closed 2025-03-06 by AlbrechtL