AlbrechtL/openwrt-docker #30: Container Startup: continuous loop

schnellert, 2025-03-07:
Hi,

I was able to solve the "Port-9000" problem. There was actually just a service on the host, which I was able to terminate.

But now I end up in one continuous loop, see screenshot:
[image]

So far I have only changed "PCI_1" in the docker-compose.yml.

Am I still missing a configuration here?

AlbrechtL, 2025-03-07:
Uh! It seems that the migration of the OpenWrt settings from the old OpenWrt image to the new one is going wrong. When you are pulling a new openwrt-docker image from dockerhub you are getting a new OpenWrt image.

I never experienced that this is not working - mh ...

Please try to delete the `openwrt-docker_data` volume and start the container again.

schnellert, 2025-03-07:
Okay.

Deleting the volume has probably brought it a bit further, now comes this:
[image]

I have a Radxa Rock5B (https://wiki.radxa.com/Rock5/hardware) with the Wireless A8 Module (https://docs.radxa.com/en/accessories/wireless-a8) and Ubuntu 22.04 (

Does the setup work everywhere?

AlbrechtL, 2025-03-07:
OK, you are the first one trying PCIe pass-through on an ARM64 based hardware :-). You definitely need the `vfio-pci `inux kernel module on your host system. If this is not included in Ubuntu you need to compile the Linux kernel yourself including that kernel module. Please also check if the RK3588 supports `IOMMU`.

schnellert, 2025-03-11:
Hi,

sorry for the late reply.

I have done a few tests.

regarding IOMMU:

`sudo dmesg | grep -i iommu`
=
[image]

`ls -lah /sys/class/iommu/`
=
[image]

and regarding vfio-pci:

`sudo lsmod | grep vfio``
=
empty.

`find /lib/modules/$(uname -r) -name "vfio*.ko"``
=
empty.

--
Does this help?

AlbrechtL, 2025-03-11:
Thanks!

You definitely need the `vfio-pci` kernel module. Can you try to update the Ubuntu to 24.04? Because on the Ubuntu x86_64 and Raspberry Pi 5 (arm64) edition this kernel module is included.

If this not helps you need to compile the kernel yourself including this kernel module.

schnellert, 2025-03-11:
Hi,

All right.

A change to 24.04 is not possible because there are no official images from Radxa yet, only community-based.

E.g. https://github.com/Joshua-Riek/ubuntu-rockchip/releases -> "ubuntu-24.04-preinstalled-server-arm64-rock5b.img.xz"

Is there a detailed instruction on how to compile the kernel with this modules? 

Thanks.

AlbrechtL, 2025-03-11:
I would recommend trying the community driven Ubuntu port.

Regarding kernel compiling: With your favorite Internet search machine you can find many generic tutorials. But there are always machine specific settings and installations steps. Because I don't know the Rockchip boards, I can not give you a recommendation. For the Raspberry Pi for example, it is documented here: https://www.raspberrypi.com/documentation/computers/linux_kernel.html

schnellert, 2025-03-11:
I have tested the Community Ubuntu 24.04 image.

`sudo cat /lib/modules/$(uname -r)/modules.builtin | grep vfio`

=
[image]

Is this a indication that vfio could work here?

The same command on ubuntu 22.04 remains empty.

schnellert, 2025-03-11:
Okay..

[image]

schnellert, 2025-03-11:
Unfortunately, that doesn't seem to work either.

AlbrechtL, 2025-03-11:
We are a step further. Can you dump the `dmesg` output, please?

schnellert, 2025-03-11:
[image]

AlbrechtL, 2025-03-11:
OK, I was hoping that there are more relevant messages. 

Is your image using `grub`? If yes, please try this: https://wiki.ubuntu.com/ARM64/performance#IOMMU_Passthrough. Important is this additional kernel argument `iommu.passthrough=1`. For u-boot it is also possible to add it and ideally there is a documentation how to do it for Rockchip.

schnellert, 2025-03-11:
Complete dmesg log, if you need it:
https://dpaste.com/A9DHX7NMS

schnellert, 2025-03-12:
Ok, I'll have to check the kernel parameters.

Another question:
Have you already been able to open an AccessPoint from wifi to wifi in OpenWRT with your setup (PCI or USB dongle)?

AlbrechtL, 2025-03-12:
>  Have you already been able to open an AccessPoint from wifi to wifi in OpenWRT with your setup (PCI or USB dongle)?

Once you successful pass-through your devices that should be possible. I did Wi-Fi access point and client but not at the same time. But I don't see why that should not work.

schnellert, 2025-03-12:
All right.

AlbrechtL, 2025-03-29:
Closed because of inactivity.

closed 2025-03-29 by AlbrechtL