AlbrechtL/openwrt-docker #31: No wireless manager

gtirth, 2025-03-27:
Hello,

I was trying to start a container and pass through usb wifi dongle but I was not able to see the wireless option under Network to be able to do anything with it. I then also treid passing throguh pcie wireless card but I still did not have no option there. I am using latest stable image.

Can you please help me figure out how can I get that option?

Thanks.

AlbrechtL, 2025-03-28:
Most likely, you need to install the drivers in OpenWrt. Can you post `lsusb` output for USB devices and `lspci` output for PCI devices, please?

gtirth, 2025-03-28:
[image]

gtirth, 2025-03-28:
please review this image logs for reference

AlbrechtL, 2025-03-28:
Thanks. OK, let's take a look...

**USB TP-Link Archer (2357:0115)**
It seems that OpenWrt doesn't have the drivers that are ready to use. Without building and patching OpenWrt, this hardware will not work.
* https://linux-hardware.org/?id=usb:2357-0115
* https://github.com/henkv1/rtw88-usb-openwrt/

**PCIe Intel Ice Lake-LP ...  (00:14.3)**
Looks better. Try to install `kmod-iwlwifi` and watch the logs via `dmesg` in OpenWrt. It could be that you need to install some firmware packages.
* https://linux-hardware.org/?id=pci:8086-34f0-8086-0074
* https://openwrt.org/packages/pkgdata/kmod-iwlwifi

gtirth, 2025-04-02:
adding the intel wifi driver seemed to work

Thanks

AlbrechtL, 2025-04-04:
Great to hear! Therefore, I'm closing this issue.

closed 2025-04-04 by AlbrechtL