AlbrechtL/openwrt-docker #32: Exited using compose file

sugoides, 2025-04-09:
Having this error using `sudo docker compose up`

```
openwrt  | nsenter: failed to execute docker: No such file or directory
openwrt  | ❯ ERROR: Status 127 while: nsenter --target 1 --uts --net --ipc --mount docker inspect -f '{{.State.Pid}}' $(cat /etc/hostname) (line 93/206)
openwrt  | ❯ ERROR: Status 127 while: PID_CONTAINTER=$(nsenter --target 1 --uts --net --ipc --mount docker inspect -f '{{.State.Pid}}' $(cat /etc/hostname)) (line 93/206)
openwrt  | 2025/04/09 06:49:31 [notice] 862394#862394: signal 15 (SIGTERM) received from 862331, exiting
openwrt  | ❯ Container is stopping. Performing cleanup...
openwrt  | Cleanup complete.
openwrt  | 2025/04/09 06:49:31 [notice] 862414#862414: exiting
openwrt  | 2025/04/09 06:49:31 [notice] 862414#862414: exit
openwrt  | 2025/04/09 06:49:31 [notice] 862394#862394: signal 17 (SIGCHLD) received from 862414
openwrt  | 2025/04/09 06:49:31 [notice] 862394#862394: worker process 862414 exited with code 0
openwrt  | 2025/04/09 06:49:31 [notice] 862394#862394: exit
openwrt  | multirun: one or more of the provided commands ended abnormally
openwrt exited with code 255
```

AlbrechtL, 2025-04-09:
Thanks for reporting.

This message is abnormal `openwrt  | nsenter: failed to execute docker: No such file or directory`.

Which OS (Docker host) do you are using?

sugoides, 2025-04-09:
OS: Ubuntu 24.04.2 LTS x86_64

sugoides, 2025-04-09:
I have used the default compose.yml

AlbrechtL, 2025-04-09:
Interesting, I'm using the same system. How did you install `docker`?

sugoides, 2025-04-09:
I think snap

sugoides, 2025-04-09:
here's my docker bin path
```
jhnnr@home-server:~$ which docker
/snap/bin/docker
```

AlbrechtL, 2025-04-09:
Ah, it's snap, this could explain the behavior.

Is it possible that you install Docker via the official repository (https://docs.docker.com/engine/install/ubuntu/) instead of snap?

sugoides, 2025-04-09:
I'm doing it right now, will update after

sugoides, 2025-04-11:
Snap was the culprit, now able to run with compose, thanks

AlbrechtL, 2025-04-11:
Perfect! Great that it is working now.

closed 2025-04-11 by AlbrechtL