AlbrechtL/openwrt-docker #34: How to increase disk size? Running out of space after installing a few packages

xiaoyao9184, 2025-04-11:
Hi, thanks for the great project!

I'm running into a disk space issue when installing `dockerd`. 
After installing just a few packages via opkg, the system quickly runs out of space and I get errors like:

```sh
opkg install dockerd
Installing dockerd (27.3.1-r3) to root...
Collected errors:
 * verify_pkg_installable: Only have 139kb available on filesystem /overlay, pkg dockerd needs 63690
 * opkg_install_cmd: Cannot install package dockerd.
```

```sh
df
Filesystem           1K-blocks      Used Available Use% Mounted on
/dev/root                 5888      5888         0 100% /rom
tmpfs                   114972      2096    112876   2% /tmp
/dev/loop0               88709     81524       139 100% /overlay
overlayfs:/overlay       88709     81524       139 100% /
/dev/vda1                16334      6390      9944  39% /boot
/dev/vda1                16334      6390      9944  39% /boot
tmpfs                      512         0       512   0% /dev
```

Questions:

Is there a recommended way to increase the available disk space (e.g., overlay or root filesystem)?

Any advice would be appreciated!

Thanks in advance.

AlbrechtL, 2025-04-11:
You are right. The default partition size is quite low.

Here is a manual way how to expand it.

1. Stop the container.
2. On your host navigate with `root` superuser into the volume e.g. `/var/lib/docker/volumes/openwrt-docker_data/_data`
3. Expand the disk image size with `dd if=/dev/zero of=squashfs-combined-<YOURUUID>.img seek=1024 obs=1MB count=0` where 1024 is the new image size in MB ([Source](https://superuser.com/questions/24838/is-it-possible-to-resize-a-qemu-disk-image)).
4. Start the container.
5. Inside OpenWrt follow [this](https://openwrt.org/docs/guide-user/advanced/expand_root) how to.
6. Reboot OpenWrt with `reboot`.
7. OpenWrt reboots a few times now.
8. Check your new disk size.
   [image]

xiaoyao9184, 2025-04-12:
Thanks for the quick insights!

Now I understand how to manually resize the disk. However, I'm currently using this project to run integration tests for an Ansible role that installs Docker. To prepare enough space for Docker installation, I’d like to automate the disk resizing process.

Based on my past experience with OpenWRT on ESXi, I know it's possible to use:

```bash
qemu-img resize -f raw openwrt.img 512M
```

Then run the script from OpenWRT documentation to expand the root filesystem: `https://openwrt.org/_export/code/docs/guide-user/advanced/expand_root`

Would it be possible to integrate this functionality into the project, and make it optionally enabled via a configuration flag or environment variable? That way, it would support use cases like automated testing environments requiring extra space.

Thanks again!

AlbrechtL, 2025-04-12:
> I’d like to automate the disk resizing process.

Understood!

> Based on my past experience with OpenWRT on ESXi, I know it's possible to use: `qemu-img resize -f raw openwrt.img 512M`

Usage of `dd` has the same purpose for RAW images `dd if=/dev/zero of=squashfs-combined-<YOURUUID>.img seek=1024 obs=1MB count=0`

> Would it be possible to integrate this functionality into the project, and make it optionally enabled via a configuration flag or environment variable? That way, it would support use cases like automated testing environments requiring extra space.

I'm currently thinking how would be the best way to add this feature. Simple possible is to add an option to automatically increase the disk size during container initialization. But the OpenWrt ext4 file system needs to be expanded, too.

So inside OpenWrt somebody has to install and call the `expand-root.sh` script. One option could to use the `OPENWRT_AFTER_BOOT_CMD` option or you do it via Ansible and ssh.
https://github.com/AlbrechtL/openwrt-docker/blob/4ca83c03aa464445f82c6c3ac4456126227da406/docker-compose.yml#L83

Would that sufficient for you?

xiaoyao9184, 2025-04-12:
Yes, exactly — as long as resizing the .img file is supported, the rest becomes much easier to automate.

AlbrechtL, 2025-04-12:
I added a new option `IMAGE_SIZE_ON_INIT` in https://github.com/AlbrechtL/openwrt-docker/commit/f30ef2f49e4c0f337b16351ea028d02c0bf27c10.

See docker-compose.yml
https://github.com/AlbrechtL/openwrt-docker/blob/f30ef2f49e4c0f337b16351ea028d02c0bf27c10/docker-compose.yml#L96-L104

Feel free to test it. You can see the pipeline run here (https://github.com/AlbrechtL/openwrt-docker/actions).

xiaoyao9184, 2025-04-13:
I've tested the resize and rootfs expansion successfully. Thank you very much for the great support!

closed 2025-04-13 by xiaoyao9184