AlbrechtL/openwrt-docker #35: Dashboard frontend limits to 3 Ethernet interfaces

dimitriblock, 2025-04-17:
As shown here, I have more than three Ethernet interfaces. the nice overview with the dashboard frontend limits to 3 Ethernet interfaces. Could the frontend element be adapted to the amount of actual interfaces?

[image]

AlbrechtL, 2025-04-17:
Unfortunately, that's a limitation by Angular Material grid list.

Possible solution which needs to be implemented: https://stackoverflow.com/questions/74376966/angular-material-grid-list-not-fitting-height-of-content

AlbrechtL, 2025-04-29:
In https://github.com/AlbrechtL/openwrt-docker/commit/9b5202d6b4ec0a9fd7b2b4efbdaab7e6a5ee8a9e I added scrollbars if the dashboard card content exceeds the available space.

[image]

Feel free to test it.

closed 2025-08-12 by AlbrechtL