AlbrechtL/openwrt-docker #37: CPU architecture not supported for x86_64 during build process

Switzerg95, 2025-05-01:
Running on a x86 system and getting this error on 22.04.5 ubuntu:

Step 21/34 : RUN echo "Building for platform '$TARGETPLATFORM'"     && if [ "$TARGETPLATFORM" = "linux/amd64" ]; then         CPU_ARCH="x86_64";         APK_EXTRA="ovmf";     elif [ "$TARGETPLATFORM" = "linux/arm64" ]; then         CPU_ARCH="aarch64";         APK_EXTRA="";     else         echo "Error: CPU architecture $TARGETPLATFORM is not supported";         exit 1;     fi     && apk add --no-cache         multirun         bash         wget         grep         qemu-system-"$CPU_ARCH"         qemu-hw-usb-host         qemu-hw-usb-redirect         nginx         nginx-mod-stream         netcat-openbsd         uuidgen         usbutils         pciutils         openssh-client         util-linux-misc         iproute2          "$APK_EXTRA"     && sed -i 's/^worker_processes.*/worker_processes 1;daemon off;/' /etc/nginx/nginx.conf
 ---> Running in a79d0640268b
Building for platform ''
Error: CPU architecture  is not supported
The command '/bin/sh -c echo "Building for platform '$TARGETPLATFORM'"     && if [ "$TARGETPLATFORM" = "linux/amd64" ]; then         CPU_ARCH="x86_64";         APK_EXTRA="ovmf";     elif [ "$TARGETPLATFORM" = "linux/arm64" ]; then         CPU_ARCH="aarch64";         APK_EXTRA="";     else         echo "Error: CPU architecture $TARGETPLATFORM is not supported";         exit 1;     fi     && apk add --no-cache         multirun         bash         wget         grep         qemu-system-"$CPU_ARCH"         qemu-hw-usb-host         qemu-hw-usb-redirect         nginx         nginx-mod-stream         netcat-openbsd         uuidgen         usbutils         pciutils         openssh-client         util-linux-misc         iproute2          "$APK_EXTRA"     && sed -i 's/^worker_processes.*/worker_processes 1;daemon off;/' /etc/nginx/nginx.conf' returned a non-zero code: 1

I checked and i am on x86 system.

AlbrechtL, 2025-05-01:
It seems that your Docker does not set the environment variable `TARGETPLATFORM`. Maybe your Docker installation is too old. Did you install Docker from the official Docker repo (https://docs.docker.com/engine/install/ubuntu/)?

Switzerg95, 2025-05-01:
It should be fairly new and other containers are running im on ver

26.1.3

Switzerg95, 2025-05-01:
@AlbrechtL 

After using the package from docker's official repo I can now complete the build process and no longer experience this error.

AlbrechtL, 2025-05-02:
Perfect!

closed 2025-05-02 by AlbrechtL