AlbrechtL/openwrt-docker #38: [Raspberry Pi 5] PCI pass-through not working. Error "vfio-pci 0001:01:00.0: probe with driver vfio-pci failed with error -22"

AlbrechtL, 2025-05-03:
By default, the Raspberry OS Linux kernel doesn't include the vfio-pci kernel module. Therefore, a custom build kernel with option `CONFIG_VFIO_PCI=m` is required. But attaching a PCIe device to the vfio-pci driver fails with `vfio-pci 0001:01:00.0: probe with driver vfio-pci failed with error -22`. Tested with kernel 6.12.22. The reason is unknown.

**Container log**
Key error message: **`ERROR: Failed to bind PCI device 0001:01:00.0 to vfio-pci. Please double check if your system supports IOMMU and if CPU virtualization features (e.g. Intel VT-d) is enabled in your bios.`**
```
openwrt  | ❯ Preparing PCI pass-through for device 0001:01:00.0 ...
openwrt  | Unbinding 0001:01:00.0 from igb
openwrt  | Loading vfio-pci kernel module
openwrt  | ❯ ERROR: Failed to bind PCI device 0001:01:00.0 to vfio-pci. Please double check if your system supports IOMMU and if CPU virtualization features (e.g. Intel VT-d) is enabled in your bios.
openwrt  | 2025/05/03 04:11:03 [notice] 1754#1754: signal 15 (SIGTERM) received from 1700, exiting
openwrt  | 2025/05/03 04:11:03 [notice] 1770#1770: exiting
openwrt  | 2025/05/03 04:11:03 [notice] 1770#1770: exit
openwrt  | ❯ Container is stopping. Performing cleanup...
```

**dmesg**
Key error message: **`vfio-pci 0001:01:00.0: probe with driver vfio-pci failed with error -22`**
```
[  112.453581] igb 0001:01:00.0: removed PHC on eth1
[  112.613308] vfio-pci 0001:01:00.0: probe with driver vfio-pci failed with error -22
[  115.264155] mempolicy: Request by 'ps' to set policy to 5 ignored
[  115.264163] mempolicy: Request by 'ps' to set policy to 3 ignored
[  122.630898] pci 0001:01:00.0: [8086:1521] type 00 class 0x020000 PCIe Endpoint
[  122.630941] pci 0001:01:00.0: BAR 0 [mem 0x1b00000000-0x1b000fffff]
[  122.631010] pci 0001:01:00.0: BAR 3 [mem 0x1b00200000-0x1b00203fff]
[  122.631323] pci 0001:01:00.0: PME# supported from D0 D3hot D3cold
[  122.631482] pci 0001:01:00.0: 4.000 Gb/s available PCIe bandwidth, limited by 5.0 GT/s PCIe x1 link at 0001:00:00.0 (capable of 16.000 Gb/s with 5.0 GT/s PCIe x4 link)
[  122.631809] pci 0001:01:00.0: BAR 0 [mem 0x1b00000000-0x1b000fffff]: assigned
[  122.631840] pci 0001:01:00.0: BAR 3 [mem 0x1b00200000-0x1b00203fff]: assigned
[  122.694070] igb 0001:01:00.0: added PHC on eth1
[  122.694113] igb 0001:01:00.0: Intel(R) Gigabit Ethernet Network Connection
[  122.694115] igb 0001:01:00.0: eth1: (PCIe:5.0Gb/s:Width x1) a0:36:9f:44:2e:c4
[  122.694511] igb 0001:01:00.0: eth1: PBA No: G15138-002
[  122.694514] igb 0001:01:00.0: Using MSI interrupts. 1 rx queue(s), 1 tx queue(s)
```

**dmesg IOMMU**
```
$ dmesg | grep -i iommu
[    0.068950] iommu: Default domain type: Translated
[    0.068953] iommu: DMA domain TLB invalidation policy: strict mode
[    0.363055] bcm2712-iommu-cache 1000005b00.iommuc: bcm2712_iommu_cache_probe
[    0.675518] bcm2712-iommu 1000005100.iommu: bcm2712_iommu_init: DEBUG_INFO = 0x20804774
[    0.676205] bcm2712-iommu 1000005100.iommu: bcm2712_iommu_probe: Success
[    0.676935] bcm2712-iommu 1000005200.iommu: bcm2712_iommu_init: DEBUG_INFO = 0x20804774
[    0.677607] bcm2712-iommu 1000005200.iommu: bcm2712_iommu_probe: Success
[    0.678328] bcm2712-iommu 1000005280.iommu: bcm2712_iommu_init: DEBUG_INFO = 0x20804774
[    0.679003] bcm2712-iommu 1000005280.iommu: bcm2712_iommu_probe: Success
[    6.606765] rpi-hevc-dec 1000800000.codec: bcm2712_iommu_of_xlate: MMU 1000005100.iommu
[    6.606774] rpi-hevc-dec 1000800000.codec: bcm2712_iommu_probe_device: MMU 1000005100.iommu
[    6.606782] rpi-hevc-dec 1000800000.codec: bcm2712_iommu_device_group: MMU 1000005100.iommu
[    6.606787] rpi-hevc-dec 1000800000.codec: Adding to iommu group 0
[    6.606791] rpi-hevc-dec 1000800000.codec: bcm2712_iommu_attach_dev: MMU 1000005100.iommu
[    6.607425] pispbe 1000880000.pisp_be: bcm2712_iommu_of_xlate: MMU 1000005100.iommu
[    6.607433] pispbe 1000880000.pisp_be: bcm2712_iommu_probe_device: MMU 1000005100.iommu
[    6.607438] pispbe 1000880000.pisp_be: bcm2712_iommu_device_group: MMU 1000005100.iommu
[    6.607443] pispbe 1000880000.pisp_be: Adding to iommu group 0
[    6.607445] pispbe 1000880000.pisp_be: bcm2712_iommu_attach_dev: MMU 1000005100.iommu
[    6.673579] vc4_hvs 107c580000.hvs: bcm2712_iommu_of_xlate: MMU 1000005200.iommu
[    6.673590] vc4_hvs 107c580000.hvs: bcm2712_iommu_probe_device: MMU 1000005200.iommu
[    6.673597] vc4_hvs 107c580000.hvs: bcm2712_iommu_device_group: MMU 1000005200.iommu
[    6.673604] vc4_hvs 107c580000.hvs: Adding to iommu group 1
[    6.673609] vc4_hvs 107c580000.hvs: bcm2712_iommu_attach_dev: MMU 1000005200.iommu
[    6.674222] vc4-drm axi:gpu: bcm2712_iommu_of_xlate: MMU 1000005200.iommu
[    6.674227] vc4-drm axi:gpu: bcm2712_iommu_probe_device: MMU 1000005200.iommu
[    6.674230] vc4-drm axi:gpu: bcm2712_iommu_device_group: MMU 1000005200.iommu
[    6.674234] vc4-drm axi:gpu: Adding to iommu group 1
[    6.674236] vc4-drm axi:gpu: bcm2712_iommu_attach_dev: MMU 1000005200.iommu
```

AlbrechtL, 2025-05-05:
I tried the no-iommu mode (kernel needs to be compiled with `CONFIG_VFIO_NOIOMMU=y`). Still no luck.

```
openwrt  | qemu-system-aarch64: -device vfio-pci,host=0001:01:00.0: vfio 0001:01:00.0: Could not open '/dev/vfio/3': No such file or directory
```
Somebody else documented the non-working `qemu` no-iommu mode here: https://stackoverflow.com/questions/79471641/qemu-vfio-noiommu-mode-error-accessing-dev-vfio-0-despite-using-noiommu-0-devi

More infos about the no-iommu mode.
* https://doc.dpdk.org/guides/linux_gsg/linux_drivers.html
* https://alokprasad7.blogspot.com/2020/10/binding-vf-to-vfio-inside-qemu.html

AlbrechtL, 2025-05-05:
I opened an issue on the Raspberry Pi OS kernel repository (https://github.com/raspberrypi/linux/issues/6834).

AlbrechtL, 2025-05-08:
Feedback from the Raspberry Pi developers: "_PCI uses `iommu-map` to describe the relationship between a particular PCI bus master (RequesterID) and how the IOMMU uses this to index into a set of page tables. The hardware in bcm2712 doesn't pay attention to this information, so you cannot virtualise separate functions - they all share the same VA domain. What you're asking for can't be supported._
Source: https://github.com/raspberrypi/linux/issues/6834#issuecomment-2862303155

Closed because a Raspberry Pi 5 doesn't support PCI pass-through.

closed 2025-05-08 by AlbrechtL

996refuse, 2025-05-15:
> I tried the no-iommu mode (kernel needs to be compiled with `CONFIG_VFIO_NOIOMMU=y`). Still no luck.
>
> ```
> openwrt  | qemu-system-aarch64: -device vfio-pci,host=0001:01:00.0: vfio 0001:01:00.0: Could [...]

hi @AlbrechtL 

the device file should be `/dev/vfio/noiommu-3` when no-iommu mode enabled.
actually, qemu doesn't support no-iommu mode by design, for some safety reason.

https://lists.gnu.org/archive/html/qemu-arm/2018-02/msg00232.html
https://github.com/996refuse/qemu-noiommu

here's a patch unaccepted by qemu community, i tried this on gpu passthrough but failed though

https://blog.74ls74.org/2025/04/20/20250420_Windows10_in_Gentoo_Linux/ (in chinese)

AlbrechtL, 2025-06-16:
@996refuse Thanks for your work! As long as `no-iommu` is not part of the mainline `qemu` there is nothing what we can do right now.

j-inner-citadel, 2025-08-08:
Hi @AlbrechtL 
Would this limitation still be an issue if LXC were used (as described in [this page](https://openwrt.org/docs/guide-user/virtualization/lxc)) instead?

AlbrechtL, 2025-08-09:
@j-inner-citadel The LXC approach is complete different from this project approach. Because LXC is using the host Linux kernel this limitation will be not existing, but you will run into other limitations because LXC is not doing a full system virtualization.