AlbrechtL/openwrt-docker #39: error: linking with cc failed: exit status: 1   |   = note:  "cc" "-m64"

ParthPrajapati-lab, 2025-05-05:
error: linking with cc failed: exit status: 1
  |
  = note:  "cc" "-m64" "<sysroot>/lib/rustlib/x86_64-unknown-linux-musl/lib/self-contained/rcrt1.o" "<sysroot>/lib/rustlib/x86_64-unknown-linux-musl/lib/self-contained/crti.o" "<sysroot>/lib/rustlib/x86_64-unknown-linux-musl/lib/self-contained/crtbeginS.o" "/tmp/rustcAqFwIZ/symbols.o" "<17 object files omitted>" "-Wl,--as-needed" "-Wl,-Bstatic" "/usr/src/qemu-backend/target/x86_64-unknown-linux-musl/release/deps/{libtide-654cdd270313568b.rlib,libasync_sse-0d0ce60b66f810ca.rlib,libpin_project_lite-f957e831531cd961.rlib,libasync_h1-993f1b7f4c854a5a.rlib,libasync_dup-e5694ba27d129b02.rlib,libasync_io-9dfcb228c418db32.rlib,librustix-65406e4bf2ca1a51.rlib,libio_lifetimes-9dbb93df1596f705.rlib,libbitflags-fe2f2b22dea850a5.rlib,liblinux_raw_sys-044355dc3a4b90b5.rlib,libpolling-dbcfb998fb9afe83.rlib,libasync_lock-aba2a932e0fc24d7.rlib,libsocket2-c1c4f784fee3de4f.rlib,libhttparse-921e54e40691f0ef.rlib,libpin_project-93453d3b6022b55c.rlib,libhttp_client-742fb240af54186c.rlib,libasync_session-621ec084a530e88a.rlib,libbincode-686e1dcdc5cd86b3.rlib,libhmac-9e74a8160331c5b5.rlib,libblake3-387151dafe5be60c.rlib,libconstant_time_eq-24579592866b6779.rlib,libcfg_if-b2cc55d35ffcdfc9.rlib,libarrayvec-3ff22e6da5f70cda.rlib,libcrypto_mac-5b727388a3e68db9.rlib,libarrayref-e64aa7caee19477e.rlib,libbase64-e8f863865ed67a54.rlib,libchrono-598d326e3ad9c300.rlib,libnum_traits-aa39f5d3ab05d789.rlib,libiana_time_zone-625e3900d1eb0044.rlib,libfemme-7b1e74a3e4eea947.rlib,libfutures_util-c32b6b50846742c2.rlib,libfutures_task-f59a1a90e1e64cf2.rlib,libroute_recognizer-2665d801e3ac193c.rlib,libhttp_types-b788e199d90fbdcb.rlib,libserde_qs-6e69f7e12e738ffb.rlib,libthiserror-d5584c0f6fbf0416.rlib,libanyhow-20ea3febb82e1ce9.rlib,libserde_urlencoded-7edd7cdc1dec27be.rlib,libserde_json-6a6f82fa477fb19c.rlib,libitoa-4a4db0109914acd0.rlib,libryu-28d65810bc68e30a.rlib,libfutures_lite-6b6f7069f60057a0.rlib,libfastrand-d0c8558c3b1d297d.rlib,libwaker_fn-dae1d8f8b1c68a47.rlib,librand-6b2eac5b9b8fd545.rlib,librand_chacha-ab3bd2141ba94be9.rlib,librand_core-71b34bb252f2a4e7.rlib,libgetrandom-62d83cb8a91396b3.rlib,libinfer-ff4b7ae7677c28b6.rlib,liburl-6da4dcc6311370c2.rlib,libidna-c2e6c75811e7c0f0.rlib,libidna_adapter-5ec2bfa336bf088c.rlib,libicu_normalizer-39e22d959097b43c.rlib,libicu_normalizer_data-5e2c3591516ecff3.rlib,libwrite16-d47b2f1392275a96.rlib,libutf8_iter-6af6e7080e8ed0b5.rlib,libutf16_iter-113219e2ba7dcd27.rlib,libsmallvec-410292eb1557ec2f.rlib,libicu_properties-05d09ec70423661f.rlib,libicu_properties_data-4200645920110e42.rlib,libicu_locid_transform-aff98c2016650f4b.rlib,libicu_locid_transform_data-c6489e09c2b4bebf.rlib,libicu_collections-63241ea066cfeb8a.rlib,libicu_provider-7767a594fe9f2c96.rlib,libicu_locid-88c9b8f9b55f0c62.rlib,liblitemap-6696de728099dc44.rlib,libtinystr-ae095f2471e10847.rlib,libzerovec-05fe4ded2c086db1.rlib,libwriteable-dd8d73f856f8316e.rlib,libyoke-c988fc9d96219a12.rlib,libzerofrom-754894b8682cc989.rlib,libstable_deref_trait-082112214cbe7c7d.rlib,libform_urlencoded-6d1dbcbaeafeac3b.rlib,libcookie-7583d98962267384.rlib,libhkdf-6e59dc8241c7fd26.rlib,libhmac-a6b300ead3f1f8eb.rlib,libcrypto_mac-4c5679dad61e9327.rlib,libsha2-3344f0bd1002a06b.rlib,libcpufeatures-4362938685199c35.rlib,libdigest-1c1dd11d1b2ae754.rlib,libblock_buffer-91f4133b6ec9cd76.rlib,libaes_gcm-42612f11ae509dcc.rlib,libghash-ccad322dfc33ea08.rlib,libpolyval-f44f7980859c3dcc.rlib,libcpuid_bool-9a935e73dac1b713.rlib,libuniversal_hash-08aa7e18cfc34987.rlib,libsubtle-2317a3ecef495641.rlib,libctr-ea342d553b7f1f12.rlib,libaes-130dd8e8060b49ed.rlib,libaes_soft-8899ae9b001cee3a.rlib,libopaque_debug-7cba123cfcd453ca.rlib,libcipher-03ab491c28d25f53.rlib,libaead-f531668207d3642c.rlib,libgeneric_array-678629fcce2c9a90.rlib,libtypenum-97cf82fd159f7c86.rlib,libbase64-4d4b69ce3efe533c.rlib,librand-7d31057a144e99e1.rlib,librand_chacha-37b6eb4465dbf5f9.rlib,libppv_lite86-d4745b2927068586.rlib,libzerocopy-fd58dd60a274f08c.rlib,librand_core-254ddc7060a45899.rlib,libgetrandom-12afdf29cf3ad7dc.rlib,libtime-44e6b14bcc933079.rlib,libtime_macros-321ccf2c789d5525.rlib,libstandback-c4c7c875fa72dd62.rlib,libpercent_encoding-e26c0a012ca0fd76.rlib,libasync_std-375dad86e00dd9c9.rlib,libmemchr-c1934c98a5fd31c8.rlib,libasync_global_executor-d22f72a21a017a29.rlib,libblocking-872c2107851bb0b2.rlib,libpiper-5d80b78543245ff3.rlib,libatomic_waker-039b06f11579f1cd.rlib,libasync_executor-5cedc25b408f1bec.rlib,libonce_cell-163c3115948a5c42.rlib,libasync_process-342e18fd564d97d3.rlib,libasync_signal-c6385be86463f556.rlib,libsignal_hook_registry-974a3949659c80d6.rlib,liblibc-f10403d93aa766d7.rlib,libasync_task-c555f1b94b45a875.rlib,libasync_channel-6cdd2d566a4cbc4f.rlib,libasync_io-fdd2b92a1e34a574.rlib,libslab-68a09765921dfe0e.rlib,libpolling-22e1a2fdbb2c6102.rlib,libtracing-dbc5ca3cf29a7a1b.rlib,libtracing_core-ef066dee076657bc.rlib,libcfg_if-7432d750e05e3f51.rlib,librustix-8ef9d0c5d5741c3e.rlib,libbitflags-95535cb6be04c994.rlib,liblinux_raw_sys-a5c09908a852e712.rlib,libfutures_lite-f75214d034e1e668.rlib,libfastrand-7ebf6cc52e249436.rlib,libfutures_io-a87ab31811bcca75.rlib,libasync_channel-d02b9e4f1a945d73.rlib,libevent_listener-2b85554929b67f72.rlib,libconcurrent_queue-094af8f4dc2b2ca1.rlib,libcrossbeam_utils-ddfbea6987e23f9a.rlib,libasync_lock-7c91af8caac06f92.rlib,libevent_listener_strategy-13172930e9fc79df.rlib,libevent_listener-71833a67626ede5f.rlib,libparking-caaaf8db9006f696.rlib,libpin_project_lite-cae75ac767bc5141.rlib,libpin_utils-fa523af34ec0bdb7.rlib,libfutures_core-7556b9e7ef46c279.rlib,libkv_log_macro-b10fc1798096fa1f.rlib,liblog-48d0275310f89570.rlib,libvalue_bag-9eb09642a1fc3e12.rlib,libvalue_bag_serde1-841c283071470e1c.rlib,libserde_fmt-7fbbd7cb0482b68f.rlib,liberased_serde-8246a45824cf9390.rlib,libtypeid-ef6543d3cc64f147.rlib,libserde-30f17d3393c31eff.rlib,liblazy_static-2e4193608c917725.rlib}.rlib" "<sysroot>/lib/rustlib/x86_64-unknown-linux-musl/lib/{libstd-*,libpanic_unwind-*,libobject-*,libmemchr-*,libaddr2line-*,libgimli-*,librustc_demangle-*,libstd_detect-*,libhashbrown-*,librustc_std_workspace_alloc-*,libminiz_oxide-*,libadler2-*,libunwind-*}.rlib" "-lunwind" "<sysroot>/lib/rustlib/x86_64-unknown-linux-musl/lib/{libcfg_if-*,liblibc-*}.rlib" "-lc" "<sysroot>/lib/rustlib/x86_64-unknown-linux-musl/lib/{liballoc-*,librustc_std_workspace_core-*,libcore-*,libcompiler_builtins-*}.rlib" "-Wl,-Bdynamic" "-Wl,--eh-frame-hdr" "-Wl,-z,noexecstack" "-nostartfiles" "-L" "<sysroot>/lib/rustlib/x86_64-unknown-linux-musl/lib/self-contained" "-L" "<sysroot>/lib/rustlib/x86_64-unknown-linux-musl/lib" "-o" "/usr/src/qemu-backend/target/x86_64-unknown-linux-musl/release/deps/qemu_openwrt_web_backend-a48bd12aa6767aac" "-Wl,--gc-sections" "-static-pie" "-Wl,-z,relro,-z,now" "-Wl,-O1" "-Wl,--strip-debug" "-nodefaultlibs" "<sysroot>/lib/rustlib/x86_64-unknown-linux-musl/lib/self-contained/crtendS.o" "<sysroot>/lib/rustlib/x86_64-unknown-linux-musl/lib/self-contained/crtn.o"
  = note: some arguments are omitted. use --verbose to show all linker arguments
  = note: cc: error: unrecognized command-line option '-m64'

error: could not compile qemu-openwrt-web-backend (bin "qemu-openwrt-web-backend") due to 1 previous error
cp: can't stat '/usr/src/qemu-backend/target/x86_64-unknown-linux-musl/release/qemu-openwrt-web-backend': No such file or directory
The command '/bin/sh -c if [ "$TARGETPLATFORM" = "linux/arm64" ]; then         cargo build --release --target aarch64-unknown-linux-musl;         cp /usr/src/qemu-backend/target/aarch64-unknown-linux-musl/release/qemu-openwrt-web-backend /usr/local/bin;     else         cargo build --release --target x86_64-unknown-linux-musl;         cp /usr/src/qemu-backend/target/x86_64-unknown-linux-musl/release/qemu-openwrt-web-backend /usr/local/bin;     fi' returned a non-zero code: 1

AlbrechtL, 2025-05-05:
Thanks for the report.

Did you do any code or Dockerfile changes?

ParthPrajapati-lab, 2025-05-05:
No, I have not done any changes in Dockerfile.
I have take this repo and enter command docker build -t openwrt-docker .

AlbrechtL, 2025-05-05:
Interesting, on my Ubuntu 24.04 with official [Docker](https://docs.docker.com/engine/install/ubuntu/) I just build it without any issues.

When you just want to use openwrt-docker without any modifications, I recommend using the ready to use images from [docker hub](https://hub.docker.com/r/albrechtloh/openwrt-docker).

ParthPrajapati-lab, 2025-05-05:
I have:
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.5 LTS
Release:        22.04
Codename:       jammy

Docker version 26.1.3, build 26.1.3-0ubuntu1~22.04.1

ParthPrajapati-lab, 2025-05-06:
Thanks a lot @AlbrechtL  
After using newest Docker version openwrt built successfully:
[+] Building 3961.0s (29/29) FINISHED                            docker:default
 => [internal] load build definition from Dockerfile                       0.2s
 => => transferring dockerfile: 10.35kB                                    0.0s
 => [internal] load metadata for docker.io/library/node:latest             5.8s
 => [internal] load metadata for docker.io/library/rust:alpine             6.7s
 => [internal] load metadata for docker.io/library/alpine:latest           5.8s
 => [internal] load .dockerignore                                          0.1s
 => => transferring context: 184B                                          0.0s
 => [build-frontend 1/7] FROM docker.io/library/node:latest@sha256:ee8a  203.0s
 => => resolve docker.io/library/node:latest@sha256:ee8a0bc5bbaece0c538c7  0.1s
 => => sha256:9e65f065d920c4ea9edbbb92c571f154751691e0cd7 2.50kB / 2.50kB  0.0s
 => => sha256:ee8a0bc5bbaece0c538c76e7c20fde6d4db319bbd5d 6.41kB / 6.41kB  0.0s
 => => sha256:1d74f212bcb59230d108ed8223ef8ca2026c0bf8cd9 6.40kB / 6.40kB  0.0s
 => => sha256:84649bff67ea459549b6f371f7045d9968d6ebf37 23.54MB / 23.54MB  9.9s
 => => sha256:de07ba6f486e0ce29760ab32d4381edabbc660a04 48.33MB / 48.33MB  9.9s
 => => sha256:48a2a14f59a002f5ef50911a0687d30beadf65bb 64.36MB / 64.36MB  24.8s
 => => sha256:0d41c7623f41e51939686bf861fe89538ae4dc 202.75MB / 202.75MB  50.1s
 => => sha256:c764c1566599d3adb873f97b74916b800275122061 3.33kB / 3.33kB  10.4s
 => => extracting sha256:de07ba6f486e0ce29760ab32d4381edabbc660a04c493e9  16.0s
 => => sha256:6d40b32381b16a378e7ec16f2304fb630a1d0a41 56.99MB / 56.99MB  48.1s
 => => sha256:ad94007e448fe478e2954e8d9f1016e75937fddbac 1.25MB / 1.25MB  25.7s
 => => sha256:39778bbc7da8e4292bd805aa52ae583ae55503fd7a69d3 445B / 445B  31.3s
 => => extracting sha256:84649bff67ea459549b6f371f7045d9968d6ebf370b815c9  5.2s
 => => extracting sha256:48a2a14f59a002f5ef50911a0687d30beadf65bbe35bde8  23.1s
 => => extracting sha256:0d41c7623f41e51939686bf861fe89538ae4dc84481ff41  75.4s
 => => extracting sha256:c764c1566599d3adb873f97b74916b80027512206157805e  0.1s
 => => extracting sha256:6d40b32381b16a378e7ec16f2304fb630a1d0a41c69fec0  29.9s
 => => extracting sha256:ad94007e448fe478e2954e8d9f1016e75937fddbacf64129  0.3s
 => => extracting sha256:39778bbc7da8e4292bd805aa52ae583ae55503fd7a69d361  0.1s
 => [internal] load build context                                          0.4s
 => => transferring context: 716.17kB                                      0.3s
 => [stage-2 1/9] FROM docker.io/library/alpine:latest@sha256:a8560b36e8  50.5s
 => => resolve docker.io/library/alpine:latest@sha256:a8560b36e8b8210634f  0.1s
 => => sha256:a8560b36e8b8210634f77d9f7f9efd7ffa463e380b7 9.22kB / 9.22kB  0.0s
 => => sha256:757d680068d77be46fd1ea20fb21db16f150468c5e7 1.02kB / 1.02kB  0.0s
 => => sha256:8d591b0b7dea080ea3be9e12ae563eebf9869168ffced1c 597B / 597B  0.0s
 => => sha256:6e771e15690e2fabf2332d3a3b744495411d6e0b00 3.99MB / 3.99MB  36.3s
 => => extracting sha256:6e771e15690e2fabf2332d3a3b744495411d6e0b00b2a  3905.6s
 => [build-backend 1/6] FROM docker.io/library/rust:alpine@sha256:661d7  167.8s
 => => resolve docker.io/library/rust:alpine@sha256:661d708cc863ce32007cf  0.2s
 => => sha256:661d708cc863ce32007cf46807a72062a80d2944a6f 2.63kB / 2.63kB  0.0s
 => => sha256:7f81e1ca5c870031b91135fe95a75437f8e6af2bb14 1.55kB / 1.55kB  0.0s
 => => sha256:324423ca17a744e2a5abf36b62747fb984f968abeff 2.55kB / 2.55kB  0.0s
 => => sha256:6e771e15690e2fabf2332d3a3b744495411d6e0b00 3.99MB / 3.99MB  36.3s
 => => extracting sha256:6e771e15690e2fabf2332d3a3b744495411d6e0b00b2aea6  1.3s
 => => sha256:ae2a8d04fc53dbb480e66cf7321229cda858c75a 59.10MB / 59.10MB  60.9s
 => => sha256:e565cca94b661582fa0c4f08b135ccee978ce 256.46MB / 256.46MB  111.4s
 => => extracting sha256:ae2a8d04fc53dbb480e66cf7321229cda858c75a9bfa488  10.8s
 => => extracting sha256:e565cca94b661582fa0c4f08b135ccee978ceff37f179c6  37.9s
 => [stage-2 2/9] RUN echo "Building for platform 'linux/arm64'"     &&  134.7s
 => [build-backend 2/6] RUN apk update &&     apk add --no-cache     mus  16.9s
 => [build-backend 3/6] WORKDIR /usr/src/qemu-backend                      0.6s
 => [build-backend 4/6] COPY ./web-backend .                               0.5s
 => [stage-2 3/9] COPY ./openwrt_additional /var/vm/openwrt_additional     0.5s
 => [build-backend 5/6] RUN if [ "linux/arm64" = "linux/arm64" ]; then     3.4s
 => [stage-2 4/9] RUN echo "Building for platform 'linux/arm64'"     &  3758.1s
 => [build-backend 6/6] RUN if [ "linux/arm64" = "linux/arm64" ]; then   959.0s
 => [build-frontend 2/7] WORKDIR /usr/local/app                            0.5s
 => [build-frontend 3/7] COPY ./web-frontend /usr/local/app/               1.1s
 => [build-frontend 4/7] RUN npm update -g npm                            34.1s
 => [build-frontend 5/7] RUN npm ci --no-audit --maxsockets 1            251.0s
 => [build-frontend 6/7] RUN npm install                                  36.1s
 => [build-frontend 7/7] RUN npm run build -- "--base-href='./'"         191.2s
 => [stage-2 5/9] COPY --from=build-backend /usr/local/bin/qemu-openwrt-w  0.4s
 => [stage-2 6/9] COPY --from=build-frontend /usr/local/app/dist/openwrt-  0.3s
 => [stage-2 7/9] COPY ./web-frontend/nginx* /var/www/                     0.2s
 => [stage-2 8/9] COPY ./src /run/                                         0.3s
[... cut ...]
nginx: [warn] conflicting server name "" on 0.0.0.0:80, ignored
nginx: [warn] conflicting server name "" on [::]:80, ignored
? Checking for KVM ...
? ERROR: KVM acceleration not detected (device file missing).
? Booting image using QEMU emulator version 9.1.2...
Could not access KVM kernel module: No such file or directory
qemu-system-aarch64: failed to initialize kvm: No such file or directory
? Container is stopping. Performing cleanup...
Cleanup complete.
multirun: one or more of the provided commands ended abnormally

ParthPrajapati-lab, 2025-05-06:
I am just clarify you i am not running docker inside any linux based PC
I am running docker inside h618 CPU based linux board

AlbrechtL, 2025-05-06:
>  After using newest Docker version openwrt built successfully:

Great! Therefor, I'm closing this issue.

For your other KVM issue, I'm created a new issue here: https://github.com/AlbrechtL/openwrt-docker/issues/40

closed 2025-05-06 by AlbrechtL