AlbrechtL/openwrt-docker #40: [Allwinner H618] KVM not working

AlbrechtL, 2025-05-06:
Successor issue of https://github.com/AlbrechtL/openwrt-docker/issues/39#issuecomment-2853474238.

Board: Allwinner H618 CPU based Linux board

Log:
Key log message: `ERROR: KVM acceleration not detected`
```
? Starting OpenWrt 24.10.1 for Docker ...
? For support visit https://github.com/AlbrechtL/openwrt-docker

**** Alpine release ****
3.21.3
NAME="Alpine Linux"
ID=alpine
VERSION_ID=3.21.3
PRETTY_NAME="Alpine Linux v3.21"
HOME_URL="https://alpinelinux.org/"
BUG_REPORT_URL="https://gitlab.alpinelinux.org/alpine/aports/-/issues"
**** End Alpine release information ****
CPU architecture: 'aarch64'
unable to initialize usb specunable to initialize libusb: -99
? ERROR: Status 1 while: lsusb >> /var/www/system_info.txt (line 45/0)
pcilib: Cannot open /proc/bus/pci
lspci: Cannot find any working access method.
? ERROR: Status 1 while: lspci >> /var/www/system_info.txt (line 51/0)
*******************************************
qemu-openwrt-web-backend proof-of-concept
*******************************************
? /storage/squashfs-combined-5a535eca-1ed0-4f55-aeca-4b76d870a7ea.img does not exist. Copying squashfs-combined-24.10.1.img to storage ...
nginx: [warn] conflicting server name "" on 0.0.0.0:80, ignored
nginx: [warn] conflicting server name "" on [::]:80, ignored
? Checking for KVM ...
? ERROR: KVM acceleration not detected (device file missing).
? Booting image using QEMU emulator version 9.1.2...
Could not access KVM kernel module: No such file or directory
qemu-system-aarch64: failed to initialize kvm: No such file or directory
? Container is stopping. Performing cleanup...
Cleanup complete.
multirun: one or more of the provided commands ended abnormally
```

AlbrechtL, 2025-05-06:
@ParthPrajapati-lab It seems that KVM is not enabled in your H618 Linux kernel. Please check your kernel configuration.

AlbrechtL, 2025-05-06:
> Can you have some user guide so i can process further because i don't what to do next???

I can't give you a general answer. It depends on the Linux system that runs on the board. Ideally, it is documented somewhere how to compile and install a custom kernel for this board. I can only give examples how other Linux distributions are doing it
* Ubuntu: https://wiki.ubuntu.com/Kernel/BuildYourOwnKernel
* Raspberry OS: https://www.raspberrypi.com/documentation/computers/linux_kernel.html
* OpenWrt: https://openwrt.org/docs/guide-developer/toolchain/use-buildsystem

ParthPrajapati-lab, 2025-05-07:
@AlbrechtL Sorry, that was my mistake—I misunderstood you. What I meant to ask is: what should I do next after building OpenWrt in Docker? I'm asking specifically about the next steps.

AlbrechtL, 2025-05-07:
I'm not sure what you mean. KVM is not working in your Allwinner H618 CPU based system. So you have to recompile your Allwinner H618 Linux kernel with KVM support. This has nothing to do with OpenWrt in Docker. Since I don't know your system, I can't give you specific steps.

ParthPrajapati-lab, 2025-05-07:
I said KVM already enable in kernel config i don't know why it not worked.

Linux Kernel Information: 
Linux lubancat 5.4.125 #1.2505stable SMP Thu May 1 16:08:00 IST 2025 aarch64 aarch64 aarch64 GNU/Linux

CPU Information:  
    Architecture:            aarch64
    CPU op-mode(s):        32-bit, 64-bit
    Byte Order:            Little Endian
    CPU(s):                  4
    On-line CPU(s) list:   0-3
    Vendor ID:               ARM
    Model name:            Cortex-A53
    Model:               4
    Thread(s) per core:  1
    Core(s) per cluster: 4
    Socket(s):           -
    Cluster(s):          1
    Stepping:            r0p4
    CPU max MHz:         1512.0000
    CPU min MHz:         480.0000
    BogoMIPS:            48.00
    Flags:               fp asimd aes pmull sha1 sha2 crc32 cpuid

AlbrechtL, 2025-05-07:
What is showing when you do `$ dmesg | grep kvm` ?

You can also check https://stackoverflow.com/questions/42754996/arm-cortex-a53-hardware-virtualization-features.

AlbrechtL, 2025-05-07:
It seems that your A53 CPU doesn't support the HYP mode, or it is not enabled. 

Just from googeling the error message `kvm [1]: HYP mode not available` I found [this](https://blog.printk.io/2016/07/u-bootlinux-and-hyp-mode-on-armv7/) article that may help you.

AlbrechtL, 2025-06-22:
Closed because of inactivity.

closed 2025-06-22 by AlbrechtL